m5stack/M5-ProductExampleCodes #24: AtomicGPS speed of movement - unit of measurement?

MacLe7400, 2020-08-24:
Hello,

I'm trying to get speed of movement of GPS module (AtomicGPS) using your example code via `GPS.s_GNRMC.TrackSpeed` parameter from `GPSAnalyse::AnaGPRMC` function in `GPSAnalyse.cpp`.

It seems this parameter returns speed values but in unknown scale.

I've compare with speed measurement with my car measurement and it seems AtomicGPS results about half of the real speed.

What is unit of measurement of this module??

Regards,
M

MacLe7400, 2020-08-26:
anyone??

vkichline, 2020-09-21:
Typically, GPS reports speed in knots. One knot = 1.852 KPH.

MacLe7400, 2020-09-25:
> Typically, GPS reports speed in knots. One knot = 1.852 KPH.

Thank you for answer. 
It seems that simple conversion knots -> kmh delivered proper results

closed 2020-09-25 by MacLe7400