m5stack/M5-ProductExampleCodes #26: 'struct i2s_config_t' has no member named 'tx_desc_auto_clear'

koenvervloesem, 2020-09-05:
Just trying to compile the Core2_Factory_test example code gives me the following error in the Arduino IDE:

```
/home/koan/M5-ProductExampleCodes/Core/M5Core2/Arduino/Core2_Factory_test/Core2_Factory_test.ino: In function 'bool InitI2SSpakerOrMic(int)':
Core2_Factory_test:304:20: error: 'struct i2s_config_t' has no member named 'tx_desc_auto_clear'
         i2s_config.tx_desc_auto_clear = true;
```

This is on Arduino 1.8.13 with the esp32 1.0.4 board support from Espressif, and the M5Core2 0.0.1 library installed.

Any pointers to what could be wrong?

vany5921, 2020-09-08:
There is no problem compiling. You can try to update the latest esp32 dependency library or comment out this sentence directly

closed 2020-09-08 by vany5921

koenvervloesem, 2020-09-08:
Thanks, I reinstalled the whole Arduino stack and now it compiles successfully.