m5stack/M5-ProductExampleCodes #27: Bala2 Example Code Not Working

RaSTuS26, 2020-10-02:
I've tried to load the Bala2 example code several times and all I get is a blank screen and no motor actions at all.

Downloaded from here:
https://github.com/m5stack/M5-ProductExampleCodes/tree/master/Application/Bala2

Using Board:
M5Stack-Core-ESP32

mrb0y, 2020-10-08:
I can compile and flash the firmware from Arduino & Platform.IO but I also have a problem with the program stopping at the "calibration finished!!!" indication.

RaSTuS26, 2020-10-08:
You're doing much better than myself. I'm using Ardiuno IDE (1.8.13), and I get nothing, just a blank screen.

mrb0y, 2020-10-08:
I can reflash using the M5Flash program for the Bala2 and it all works but I would like to be able to build from source.

RaSTuS26, 2020-10-08:
Yeah, I saved the original FW with esptool.py and can reload that no problems. But I'd like to compile from source also.

mlde50, 2020-10-09:
I have the exact same problem with Arduino or platformIO. I was forced to use easyLoader app to come back default Bala2 example.

mrb0y, 2020-11-01:
RaSTuS26 the problem is you are not using the M5Stack-Grey. I used the following build environment in PlatformIO and everything builds and works correctly. It is possible when I had the problem that I had previously chosen the wrong "board" for the build target.

platformio.ini
[env:m5stack-grey]
platform = espressif32
board = m5stack-grey
framework = arduino
lib_deps = m5stack/M5Stack@^0.3.1

RaSTuS26, 2020-11-01:
> RaSTuS26 the problem is you are not using the M5Stack-Grey. I used the following build environment in PlatformIO and everything builds and works correctly. It is possible when I had the problem that I had previously chosen the wrong "board" for the build target. I'm considering this a closed issue as far as I am concerned.
> 
> platformio.ini
> [env:m5stack-grey]
> platform = espressif32
> board = m5stack-grey
> framework = arduino
> lib_deps = m5stack/M5Stack@^0.3.1

There's no such board in Arduino IDE, only "M5Stack-Core-ESP32" which is described in M5 docs the one to use for the "Grey"
.

I stopped using PlatformIO ages ago, I was having more problems than using it was worth on Linux.

RaSTuS26, 2020-11-03:
Eureka, got it.

```
M5Stack Grey Arduino Settings;
 - Board:                     "M5Stack-Core-ESP32"
 - Flash Frequency:    "80MHz"
 - Flash Mode:            "QIO"
 - Partition Scheme:   "No OTA(Large APP)"
```

Most of these are default settings, the "Partition Scheme" is the most important though.

closed 2020-11-03 by RaSTuS26