m5stack/M5-ProductExampleCodes #28: Core2 Factory Test PSRAM malloc failed

kd8bxp, 2020-10-11:
Sketch is too large when compiled with M5Stick Core on Arduino IDE 1.8.13 (should a different core be used?)
[image]
[image]

Changing "partition scheme" to No OTA large app allows the sketch to compile and upload

The device gives a PSRAM malloc failed error on the screen, tapping the screen will get past it.
but the screen doesn't display correctly.
touching the imu block, doesn't update the screen, or the imu.

felmue, 2020-10-14:
Hi @kd8bxp 

I don't know if it exists in Arduino IDE (as I am not using it), but in PlatformIO I always select `M5Stack FIRE` since that Core has PSRAM as well.

Cheers
Felix

kd8bxp, 2020-10-14:
Hi @felmue yes the FIRE core is in Arduino, and that solved the problem. I didn't even think about using it, I guess the word "core" made me think it was a M5Stack Core - should  have known better.
I really need to learn to use PlatformIO The Arduino IDE drives me nuts at times.  Know of any good tutorials? 
I am closing this issue, thanks for pointing me in the right direction, and for the help.

closed 2020-10-14 by kd8bxp

felmue, 2020-10-14:
Hi @kd8bxp 

glad to hear it worked - enjoy your Core2.

It's been a while since I've started with PlatformIO, but I think I simply followed the VSCode/PlatformIO installation process here: https://docs.platformio.org/en/latest/integration/ide/vscode.html

And yes, you most likely will not regret moving on to PlatformIO. I didn't and I never looked back.

Happy Stacking!
Felix