m5stack/M5-ProductExampleCodes #41: CAM2CORE not working?

TheAnton205, 2021-07-28:
Hello, I am using [unit cam](https://docs.m5stack.com/en/unit/unit_cam) with a Core 2. I want to communicate through UART, just like the video on that page. I followed the connections given, and uploaded the arduino sketch. After doing so, the blue LED blinks for three times then my previous firmware completely freezes and nothing else works. I have attempted to to erase the previous firmware, then uploading the arduino sketch. Nothing.

I have tried to use the m5flow programming blocks, yet there is no support for the unit? I attempt to click on it, yet it is grayed out. [See this image.](https://ibb.co/xfyqs92)

I don't understand, am I doing something wrong?

felmue, 2021-07-29:
Hello @TheAnton205 

from what I can tell the `unit cam` Arduino example and UIFlow integration only is available for M5Stack (Base, Fire, etc.) but not yet for M5Core2. (BTW: That is the reason `unit cam` is grayed out in UIFlow when M5Core2 is selected.)

At this point I see three options:
- wait until M5Stack engineers have adapted `unit cam` for M5Core2.
- try to adapt the M5Stack Arduino example for M5Core2.
- get an M5Stack (Base, Fire, etc.) and a M5Go base (UIFlow uses port B or C).

Thanks
Felix

veryalien, 2022-06-26:
I've just realised that this suggestion is relatively simple:

try to adapt the M5Stack Arduino example for M5Core2.

You can simply replace the first line of the m5stack arduino sketch cam2core: #include <M5Stack.h>
with: #include <M5Core2.h>

[https://github.com/m5stack/M5-ProductExampleCodes/tree/master/Unit/UNIT_CAM/CAM2CORE](https://github.com/m5stack/M5-ProductExampleCodes/tree/master/Unit/UNIT_CAM/CAM2CORE)

Compile and upload the sketch to a core2 and connect a unitcam to Port B (26, 36) and it should display the camera video stream on the screen.