m5stack/M5-ProductExampleCodes #51: Unit-Key_noBlock.m5f is not m5f but HTML

3110, 2022-07-19:
https://github.com/m5stack/M5-ProductExampleCodes/blob/master/Unit/KEY/UIFLOW/Unit-Key_noBlock.m5f

The UiFlow sample of Unit Key is not the m5f file but the HTML file.

[image]

IcingTomato, 2022-07-19:
You can right click and "Save as..." to save as *.m5f file.

3110, 2022-07-19:
If you mean the UiFlow sample of Unit Key is the same as Unit Limit, you shoud commit the following JSON file as m5f file not HTML file of the GitHub page.

https://raw.githubusercontent.com/m5stack/M5-ProductExampleCodes/master/Unit/LIMIT/UIFlow/Unit-Limit_noBlock.m5f

Tinyu-Zhao, 2022-07-19:
> If you mean the UiFlow sample of Unit Key is the same as Unit Limit, you shoud commit the following JSON file as m5f file not HTML file of the GitHub page.
> 
> https://raw.githubusercontent.com/m5stack/M5-ProductExampleCodes/master/Unit/LIMIT/UIFlow/Unit-Limit_noBlock.m5f

GitHub doesn't support .m5f format files, not that I want him to be in HTML format, and the content of the .m5f format file itself is such that you can try keeping a .m5f on UIFLOW yourself

IcingTomato, 2022-07-19:
I will upload a new one from our FAE.